amchagas/Flypi #6: h264 conversion

amchagas, 2016-02-15:
Right now we record h264 videos, and to be able to read them into imagej/Fiji, we have to convert it to mp4, then to avi, which is a bit idiotic. Specially because the MP4 to avi conversion is done in a tool that is not available for linux.

grg2rsr, 2016-11-08:
Hi,

here is a code snippet converting h264 into imagej importable .avi with the mjpeg codec using `avconv` (which is basically `ffmpeg`), included in `libav-tools`
```
import subprocess
import sys
import os

# find data filenames
def find_files(directory,ext):
    """ returns all files with the file extension 'ext' in a directory
    tree starting at node 'directory' """
    data_fnames = []
    for dirpath, dirnames, filenames in os.walk(directory):
        for filename in filenames:
            if filename.endswith(ext):
                data_fnames.append(os.path.join(dirpath,filename))
    return data_fnames

def convert(fname,fmt,br,codec):
    """ converts to format 'fmt' with bitrate 'br' and codec 'codec' """
    outname = os.path.splitext(fname)[0]+'.'+fmt
    command = ['avconv','-i',fname,'-b:v',br,'-codec',codec,outname]
    subprocess.call(command,shell=False)

files = find_files(sys.argv[1],'h264')    
for fname in files:
    convert(fname,fmt='avi',br='2M',codec='mjpeg')
```
great work on the flypi. Enjoying it Tansania right now :)
cheers, Georg

amchagas, 2016-11-13:
Hi! 
sorry just saw that you had a different bit of code here... My bad!

amchagas, 2016-11-13:
@TomBaden,
Just updated the master branch to the newest code I have.
Implemented there should be:
-Horizontal and Vertical offsets now start in the middle of the frame.
-AVI conversion seems to work (didn'T check quality after conversion in usefull software)
    -- Needs "avconv" to run: "sudo apt-get install libav-tools"
-Protocol should be working for all classes except peltier

Let me know how testing goes?

closed 2017-05-22 by amchagas