amchagas/Flypi #8: USB connectors and inductors footprint

ifriad, 2016-09-16:
It will be very useful if the connectors to the ring, fan, etc. were changed to USB_A. This will make it easier for users to plug in/out external modules at their convention.

Please also take care of the inductors footprint on the PCB, they are of the wrong size.

amchagas, 2018-06-22:
Hi Ihab,
we solved these and other issues on the Prometheus fork where development is continuing https://github.com/prometheus-science/Flypi

closed 2018-06-22 by amchagas

amchagas, 2019-11-17:
@all-contributors please add @ifriad for design, code

amchagas, 2019-11-17:
@all-contributors please add @GrandeCappucino for design, code

amchagas, 2019-11-17:
@all-contributors please add @TomBaden  for design, code

amchagas, 2019-11-17:
@all-contributors please add @GrandeCappuccino for design

amchagas, 2019-11-17:
@all-contributors please add @icaven  for design

amchagas, 2019-11-17:
l-contributors please add @amchagas  for design, code, documentation

amchagas, 2019-11-17:
@all-contributors please add @amchagas for design, code, documentation