amchagas/Flypi #31: Agar plate imaging and colony counting module #MiPi:zap:

vektorious, 2018-09-16:
This issue is part of the [MiPi:zap: extension](30)! 

**We want to develop an agar plate imaging and live colony counting system!**
Because colony counting will be relying on image analysis algorithms image quality is a key factor for a low error rate. That's why a good agar plate imaging system is essential for this and all future MiPi:zap: modules.

**Experience with imaging Agar plates:**
Problems: 
- reflections in the agar (wet surface!)
- imaging through the agar or from the top  impacts image quality
- possible contamination
- different bacteria behave differently -> low consistency regarding colony opacity

Experience/Solutions to avoid the Problems:
- indirect lighting of the agar plate produces high contrast images (pseudo "dark field", [other project](https://github.com/vektorious/mbac/issues/10))
- uniform lighting through a [white weighing dish](https://images-na.ssl-images-amazon.com/images/I/71WFcb8HNqL._SX522_.jpg) might be good enough!
- a good algorithm can compensate problems with image quality/different colony opacity (e.g. [OpenCFU](http://opencfu.sourceforge.net/))

First steps:
- test lighting setup with already existing modules of the FlyPi
- develop algorithm using example images

qgeissmann, 2018-09-16:
Hey, I am the developer of OpenCFU and [ethocopes](https://github.com/gilestrolab/ethoscope). I am interested in your project. I am moving towards plant-bacteria interactions for my postdoc so will be interested in contributing/collaborating.

vektorious, 2018-09-16:
Welcome @qgeissmann ! That's great!

Getting OpenCFU integrated into the FlyPi would avoid reinventing the wheel but until now the FlyPi software is mainly based on python. What do you thin, would a scaled down version be fast enough for live analysis in a camera stream? Maybe with reduced FPS?

qgeissmann, 2018-09-16:
It would be definitely possible to reimplement it in python. It is mostly OpenCV based anyway. I would not aim for true real-time counting though. I think in practice, you get the best image you can and then the algo would be able to cope with it from a still picture. Alternatively, I wanted to develop a tensorflow based counting/imaging platform in the future... not sure I will have the time though...

vektorious, 2018-09-17:
Perfect! Well we also discussed about using Tensor flow but we don't have much experience with computer vision and we are happy about every input we can get!

At the moment we are preparing things to work/discuss about the MiPi extensions at the ["Gathering for open science hardware meeting"](http://openhardware.science/gatherings/gosh-2018-2/). If you haven't yet, consider joining the [GOSH forum](https://forum.openhardware.science/)!

amchagas, 2019-08-14:
Hi @vektorious!
as far as I understood you guys moved this development to a whole new place/repo, right? 
Would you mind sharing the link here so that we can close this issue?

vektorious, 2019-08-17:
Well actually not really. We just changed the name from mbac to lightM and therefore we also changed the repository. I edited the first post and added the new link. Unfortunately there is not much progress on this project.

But there is another project which might also produce some useful modules which could be used to realize this FlyPi extension.