amchagas/Flypi #34: Make PCB Modular

amchagas, 2019-02-03:
Currently the system has 1 PCB containing all hardware modules. Optimally, each module would be implemented in a small PCB, as a piece of a puzzle. Seeed has implemented something like this with their [grove system](https://www.seeedstudio.com/category/Grove-c-1003.html). 

The advantage of this modularity would be to reduce costs (as users only need to source parts for the modules they need and each PCB will be smaller). Also it will allow bigger customization as users might want 3 LED modules and no Peltier module (as opposed to the stock 2 LEDs and 1 Peltier).

isobianin, 2019-09-23:
Hi! So I addressed this issue later and came with a plan of solving it. Nothing special but it will help with keeping this issue transparent to those who are interested in it and it just keeps everything organized. Additionally, it may help with estimating time needed to achieve desired result.
Plan of work:
1. Look into schematics and PCB layout.
2. Examine analogous systems.
3. Work on modularity.
3.1. Design new PCB layout.
3.2. Check PCB signal integrity and electromagnetic compatibility (Verify PCB's ability to function in general).
In order to accomplish set goals I would like to know the following:
1) This issue was posted a long (relatively) time ago. Are there any time restrictions? 
2) How is PCB assembled and manufactured? I suppose that you order it in assembled state from somewhere.
3) What components are available? There should be a little database where one can find required connectors and other components.
4) Even so I had looked through the folder "PCB", I got confused a little bit because there are so much files. Can you please explain what are the files that I can work with? I mean, what are the files for latest PCB schematic, layout and bill of materials?

amchagas, 2019-09-30:
Hi!
Thanks for putting some time and effort in thinking about this. 
Here are answers in order:
1 - no time restrictions, this has been a plan when this project was more active, but it went to the "back burner" as other priorities popped up.
2 - Never ordered it as an assembled piece from anywhere. we mostly get PCBs from pcbway or jlcpcb and use kitspace.org to get components. https://kitspace.org/boards/github.com/amchagas/flypi/
3 - all the components are here https://github.com/amchagas/Flypi/blob/master/PCB/flypi_pcb_kicad/1-click-bom.csv
4 - here are the latest files: https://github.com/amchagas/Flypi/tree/master/PCB/flypi_pcb_kicad

isobianin, 2019-10-01:
Hi! Okay, thanks. This information sorts some things out. I will try to post here updates on current progress and relevant questions

isobianin, 2019-10-21:
Hello @amchagas !
Sorry for the delay, here is an update. It contains some questions and propositions on the future design.

At first, I have found some info on this PCB in general (fig. 1). 
[image]
Figure 1 - FlyPi structure
It consists of:
a) power 1 - 12v to 5v converter dedicated to power the Raspberry Pi
b) power 2 - 12v to 5v converter to power LEDs, LED ring, Thermistor and devices on the I2C port
c) High power LED / 5V general module 1 
d) High power LED / 5V general module 2 
e) LED ring module 
f) servo motor module 
g) Peltier element module 
*) also there is SD card (I2C port)

The question is whether should all this stuff be modular? It is possible to do each module but I suppose things like power converter a) and b) will always be on this board. On the other hand, there are LEDs, servo motor and Peltier element module. LED ring is represented by one small plastic socket so I do not think that it is a problem to leave it there. 

Then there is a question of how can modularity be implemented? I see two main ways:
1) small PCBs are connected to main PCB via pin headers. This way the end result will look like mother board
2) small PCBs are connected to main PCB via wires through plastic sockets (just like LEDs and other instruments are connected to PCB now but new elements will be added to the current structure)

The first design may have different approaches:
- PCBs are placed perpendicularly to the main PCB (fig. 2)
[image]
Figure 2 - Perpendicular
pros: approach uses Z-axis and allows for more things to placed
cons: PCB enclosure is very tight and PCB may have problems with dimensions (for now, I have checked that LEDs can be implemented that way but more research is needed)
- PCBs are placed on top of the main PCB (fig. 3)
[image]
Figure 3 - On top
Pros: straightforward, relatively easy design
Cons: approach creates a lot of unused space below modular PCBs
- PCBs are parallel to each other (fig. 4)
[image]
Figure 4 - Parallel
Pros: relatively easy design
Cons: PCB will be cut, best used only on border modules (power, peltier), LEDs – not so much

Of course, it can be mixed. For example, LEDs are placed perpendicularly, big modules like power converters and Peltier are connected in parallel.
The second design probably should have some form of an enclosure. It may look like a box that have shelfs for PCBs (fig. 5).
[image]
Figure 5 - Enclosure

There are some questions on schematics:
1) there is a small discrepancy and capacitors C1 and C2 are mixed up
2) why is Arduino powered from CON1 Barrel_jack directly? There are no converters and 12 V (presumably) connect to the 5V pin.
3) maybe, if we are to make some changes in layout, we may increase the total number of LEDs that can be connected? I see that there are some pins that may do the thing. Also, we can add additional register and have even more LEDs

My vision of the solution is the first approach with pins and make it solely perpendicular (fig. 6). Try to do it this way or make it mixed.
[image]
Figure 6 - Design
What are your thoughts on amount of modules, design approaches and schematics?

P.s. There is a question in the issue #47. I have created the pull request because I have not recieved an answer yet

amchagas, 2019-10-23:
Hi @isobianin, very sorry for the long delay!!! 
I was travelling for work these days.
I like design 1 (figure 2) and variant B. You are also right to say that some things will always be there, such as the power modules on the right of the board. 
1 - I don't quite get what you mean with the capacitor discrepancy. Do you mind trying to explain it?
2 - The arduino is powered this way, because that is a volt_in port. And from the arduino datasheet, it can take up to 20V. (check power section on documentation tab here https://store.arduino.cc/arduino-nano). This was done this way to avoid draining too much power via the raspberry pi (as the other option would be for the arduino to be powered by USB)
3 - I wonder if we would be better thinking about this ports/spaces for small PCBs not as leds, but as "general" ports? I mean, the main PCB would have "ports/pcb connector" that would be three pins basically power/ground and data. and then each smaller pcb could be designed/configured for whatever function, including new things? In that way this would also be flexible enough to simply add cables if wanted/needed instead of the small PCBs?

isobianin, 2019-10-24:
Hi @amchagas, no problem, hope the business trip was successful.
1 - It is that identification numbers of capacitors in the schematics and in the layout are mixed up (fig. 1). It is a small detail that I had noticed and were puzzled at first.
[image]
Figure 1 - Discrepancy in components
2 - Got it, very informative. I knew that there was something with the pins' specification. 
3 - Totally agree with the first part of the paragraph. I was about to propose something like that because it is practical to design universal module that can be easily programmed. It just need to have enough pins. However, I did not quite understand the other part of the message. So are you referring to the second design where the main board consists of plastic connectors (or pin headers) which are connected to anything with the corresponding pins? In this case, what about driving circuits (transistors)? I mean there are examples without drivers on the same main board.
[image]
Figure 2 - Drivers
I came up with the idea of implementation of small PCBs based on an assumption that all the stuff on the main board is essential for its connection to the desired object (led, servo motor, led ring etc.). If the desired object resembles complete module (with all necessary driving elements), it is possible to replace small PCBs with connectors. Nevertheless, peltier will still remain as the one big module that, in all likelihood, will be turned in to the PCB.
Here I drew general concepts (fig. 3).
[image]
Figure 3 - Concepts
The A concept is what I was going for. The main issue of the B concept is the driving circuit. If it can be omitted, then it is settled and general-purpose connectors will be inserted on the main PCB.

isobianin, 2019-10-29:
Hello @amchagas! I see that we are still in phase of discussion of how general ports should look (small PCB or just ports). Therefore, while we are deciding what is better, I want to start designing the separate PCB for Peltier module due to the fact that we have already agreed on this particular module design. My algorithm is following: I will try to connect this module and main module via 90° pin headers. If the size of the designed PCB is too big for the enclosure and the main board, I will report it and discuss possible solutions

amchagas, 2019-10-29:
Hi @isobianin ,

1- Thanks for spotting that! 

about the last point, I was just wondering if we should think about a more general solution. Exactly in the lines of what you proposed with Concept A and B. Or like I mentioned in the first topic of this thread, like the grove connectors from Seeed studio. So in this way the main board would only have the power supplies and the arduino board. 
The rest would be a system/connector(s) layout for people to connect small PCBs containing different circuits? I don't know yet if this makes sense, given the many breakout systems that are already available out there.... Also I don't know how much sense would it make to keep changing this FlyPi design. Would it make more sense to keep this one as is, and move on to a new repo/project where we can have more freedom to test things? For instance, why not use an ESP32, that has more connection ports or even wifi, so that we have a more general system for different applications?

isobianin, 2019-10-29:
Regarding breakout systems, I think that we need to identify what makes FlyPi uniqe. What I understood during the given period is that it is more like DYI thing with through-hole elements and with a set number of modules, thus making it a bit cheaper alternative to, say, the Grove system. I would also point out that FlyPi is designed specifically for microbiology. This way it is more preferred by researchers to buy FlyPi than general-purpose PCBs. Probably, there is more to it but that is all I can say for now.

Speaking about new project, yes, this is definitely a better way of upgrading FlyPi and creating something more than it is now. New microcontroller would be a nice addition.

My opinion is that FlyPi is an existing project. If it exists then it is needed in some specific research areas. Therefore, an upgrade is likely to be a welcome addition

isobianin, 2019-11-04:
Hi @amchagas ! I spent some time thinking about the completely new layout of PCB we had discussed earlier. Probably, the best case would be to implement modularity, as you suggested, via small PCBs connected through cables. This way we can create the small set of PCBs that is specifically design for FlyPi project subject area (microbiology) so people are able to use them with their current breakout system (fig. 1). If they do not have one yet, that is when we may offer them the enhanced version of FlyPi PCB.
[image]
Figure 1 - The concept of designated modules
Here, the main objective is to determine specific modules and modules that are to be implemented in the future. I mean the current version of FlyPi have a lot of things for research purposes. However, there should be some ideas that were omitted due to the lack of manpower or time (maybe both). 

To summarize:
1) What about this concept? The small set of specific purpose modules.
2) What about specific modules for FlyPi? Is it Peltier module or there is more?
3) What modules are needed and we can develop? I see it as the complex CAD/electronics issue and interesting task.  
4) If we are good to go, what can I start working with?

amchagas, 2019-11-07:
Hi @isobianin !

Thanks for putting time into thinking about this system. In principple I like the idea. One thing that is not entirely clear is what do you mean by "breakout system"? I mean, how does it differ from the specific purpose module, or from the controlled object?

In terms of controlled objects, here are some candidates for research in general:
- Servo motor (depending on size needs a dedicated power source, where the 5V output from arduino won't cut it)
- High power LEDs (dedicated power supply with high power resistor to regulate current?)
- Peltier element (h-bridge)
- transistor switches (kind of a "generic" system to turn things on/off)
- Simple Vcc, signal, ground  breakout
- temperature sensor
- humidity sensor
- gas sensor

I think this is could be generic and big enough to be transformed in a different project. This would be better for people who read the paper describing the Flypi, as they would still find this version which is somewhat similar to what is described in the paper, and better for us as it would allow us to develop without previous "prejudices" and constraints.

isobianin, 2019-11-07:
Hi @amchagas !
By the breakout system I mean some kind of main PCB (the hub perhaps?) which consists of the microcontroller and lots of ports. These ports are connected to specific modules, for example Peltier. Then the module is connected to things it is supposed to control. Of course, specific module may have controlled object on board (like sensors).

Speaking about modules, I will think this through as much as I can.

Totally agree on all points about the new project.

amchagas, 2019-11-11:
Hi @isobianin,
cool, I get it now. 
About the new project, would you like to setup a Github repository and invite me as a collaborator? 
If you are not that confortable with Github yet, I can also do this. Just let me know!

isobianin, 2019-11-14:
Hi @amchagas! Sorry for the delay, I created the new repository and invited you

closed 2021-05-27 by amchagas