amchagas/Flypi #36: Camera feed into a File/thread/buffer

amchagas, 2019-02-03:
Currently the PyCamera library sends the image to the screen sending data directly to the GPU, and bypassing the Pi's CPU:
> That takes care of how each line’s read-out time is kept constant, but it still doesn’t answer the question of how we can guarantee that Linux is actually listening and ready to accept each line of data? The answer is quite simply that Linux doesn’t. The CPU doesn’t talk to the camera directly. In fact, none of the camera processing occurs on the CPU (running Linux) at all. Instead, it is done on the Pi’s GPU (VideoCore IV) which is running its own real-time OS (VCOS).

This means that connecting to the PI via SSH doesn't allow for the camera feed to be displayed on the remote computer. We would like to solve this issue by sending the camera stream to eg. a file, and then sending the data via SSH. This has already been implemented by the [waterscope project](https://www.waterscope.org/), so that would be a good place to start looking.

Important here would be to know:
- what is the lag this generates ( can users focus their samples using only the feed sent via SSH? This way we could bypass needing a keyboard, mice and computer screen).
- What would be a safe way to implement this for true remote access (Flypi in Germany, accessed from the UK). [Mozilla IoT](https://iot.mozilla.org/) project might be a place to start looking.

amchagas, 2020-10-01:
@isobianin related to our discussion yesterday, and one of the issues I mentioned.

I think this code from the PiCamera class should work for what is needed 
https://picamera.readthedocs.io/en/release-1.13/recipes1.html#recording-to-a-network-stream

The main question is how much latency this would add, since if there is a lag between samples being moved under the microscope and actually seeing the change in the screen makes it hard to work with the device.

isobianin, 2020-10-20:
Hi @amchagas , I have done some research in this field and there are some information and questions. Hope, that you can correct me if I am wrong somewhere.

First of all, I have looked up the links provided. The first question is whether do we really not use SSH? What I mean is that practically all, if not all, guides on establishing client-server streaming with RaspPi require enabling SSH via its settings. 

Speaking about streaming, I have noticed that there is a quite huge latency added. However, there is a way to increase FPS/decrease latency via so-called [threading](https://www.pyimagesearch.com/2015/12/28/increasing-raspberry-pi-fps-with-python-and-opencv/) (from what I could fathom, it is a parallel or pseudo-parallel computing). Threading is described in PiCamera [documentation](https://picamera.readthedocs.io/en/release-1.3/recipes2.html#rapid-capture-and-streaming), too. With this in mind, am I correct in an assumption that, _firstly_, we need to establish connection between RaspPi and then, _secondly_, think about an increase in FPS?

Lastly, I am thinking about testing. Particularly, how can we test all of these? And how can I safely test everything on my side before deploying it here?

amchagas, 2020-10-22:
Hi @isobianin, 
Thanks for looking into this. 

The way I see it, there are two different issues "mixing up" here. 

- One is to connect to the Rasp. via SSH so that is can be controlled without the need to be physically next to the device, and to avoid the need to have it constantly working with a keyboard, mouse, monitor.

- the other is how the video information is transmitted from the camera. In "regular" use, using the Picamera library, the data from the camera is displayed straight through the attached computer screen. But if anyone would like to access the Pi remotely (either using SSH and its graphics relaying option, or VNC), they would not  be able to see the camera data. This second issue is what I think needs work on, as once this is done, remote access should be straight forward. 

As you mention, it could be that sending the camera data to a buffer/thread/network port leads to a lag that is too long and the device becomes unusable. I guess the only way to find out is testing it...

Why don't we do it like this: If it is fine for you, I can send you a kit for a Flypi. This way you can have all the needed things to actually test/play around with things.  If that is fine, send me a post address via email?

isobianin, 2020-11-26:
Hi @amchagas !

There is a an update regarding connection. I was able to establish a client-server relationship between FlyPi and a laptop respectively. The code utilizes the latest additions from the picamera documentations, particularly [rapid capture and processing](rapid-capture-and-processing) on a client side and OpenCV on server side. The video is converted to MJPEG as I found that it is the fastest way to transfer it. OpenCV is used for quick and effective deciphering of the input image from the camera.

Regarding the latency, the first test launch showed 28 fps which is quite impressive as the image is not that compressed. However, when I tried it again after sometime and the fps dropped dramatically to 1-6 frames. I believe that this may be caused by my local WiFi network that sometimes stutter, as after some time of consequent launches it again showed 20 fps mark. Maybe RaspPi got too heated up after all this testing, I cannot really tell.

The files are [here, named server-rapid and client-rapid](https://github.com/isobianin/Flypi/tree/master/Python).

isobianin, 2020-11-29:
Hi @amchagas !

I have updated the server script as I found out that a wrong file was uploaded. Should work fine now

isobianin, 2020-12-01:
Hi @amchagas !

I wanted to try out alternative way of streaming and there are some good results too (very similar to the obtained above). They are based on this [section](https://raspberry-projects.com/pi/pi-hardware/raspberry-pi-camera/streaming-video-using-vlc-player).

For this we will need VLC player on server and raspivid on RapsPi client. Frankly, raspivid was already installed on the image you sent me. The setup for streaming consists of **two steps**.
**Step 1**. In RaspPi console we type in:

raspivid -o - -t 0 -n -w 600 -h 400 -fps 30 | cvlc -vvv stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/}' :demux=h264

It will have about 1-2 second delay but the latency is quite low. 
**Step 2**. In VLC player we will need to input the following:

rtsp://###.###.###.###:8554/ with the respective raspberry pi address

[Finding RaspPi IP](https://www.raspberrypistarterkits.com/how-to/find-raspberry-pi-ip-address/)

I believe that these both methods need further testing on site

amchagas, 2020-12-01:
hi @isobianin ! 
Thanks a lot for the work on these! 
I'll take a look at it the next few days, but seems from what you describe that these are quite stable solutions :+1:

amchagas, 2020-12-04:
This looks great as well! 

I like the more pythonic solution as this will allow for the code to be included in the GUI and allow users to start the client from the GUI. 

I think the ultimate package for this issue would be:

- Have a button on the GUI to start the client-rapid script (maybe a text box to add the IP address - but not essential) - This way we can use the GUI camera controls to adjust brightness, contrast etc.
- Have the script only preview the video, instead of necessarily recording data (so start_preview instead of start_recording), as this way we can send data for an indefinite amount of time without taking space resources

Let me know if you have time to look into this, or if we should just do a PR for the current code and focus on the items I mentioned later

Again, thanks for working on this!

isobianin, 2020-12-04:
Hi @amchagas! 
I guess I can look into this. This issue is quite interesting and I am glad to hear that it works fine. Also, I got somewhat competent at this so it is only fitting to get it done the best way. I will tinker with the issue some more

isobianin, 2020-12-05:
Hi @amchagas !

I do not exactly understand what you mean in your second point. Does it mean that the Pi stops the stream automatically after sometime due to the stack/buffer overflow? This is important as the client-side script implements a method _flush()_ which should reset the file. I have spent sometime reading the documentation and it I think it is how it works. I also run a test for 5 minutes and got good results - 26.32 fps in 5 minutes and I saw no additional memory used (I checked it via df -h) before, during (checked pretty much constantly) and after the test.

amchagas, 2020-12-06:
Hi!
Oh never mind, I only read over the client.py file and thought it was generating a file everytime it ran, which is not the case. 
One thing that we might need to deal with though, is that when it is integrated on the GUI and users actually want to record videos or take photos with the camera.. My question is if using "camera.start_recording(output, format='mjpeg')" we are still able to start another recording. or if this function "blocks" the camera

isobianin, 2020-12-07:
Hi @amchagas !

I see. I thought about this too as it would be a logical thing to do to record and stream simultaneously. I am working on this right now as it is quite hard to get it running properly.

As a side not, what I was able to do for now is to indeed establish a stream from GUI with custom IP and PORT while utilizing _resolution_ parameter from the drop menu. So, from what I was able to understand, the lower the _resolution_, the higher FPS will be but it is possible to stream with any _resolution_. I am trying to make it all work in parallel as now the stream freezes the GUI. Actually, the previously implemented "record" function freezes GUI too, but it is not as noticeable as it does not have to run for so long.

isobianin, 2020-12-09:
Hi @amchagas !

I have made some improvements on the code and it works good. In essence, I was able to utilize the threading functionality thus making possible to operate with FlyPi while steaming. Moreover, now it is possible to simultaneously record videos, stream and work with other peripherals (led, peltier, etc.). The GUI is simple with just fields to enter an IP address and a PORT (if you clear a field, you will see xxx.xxx.xxx.xxx written). The default IP address and PORT can be changed in flypi_GUI.py (fig. 1).

[image]
Fig. 1 - GUI with streaming functionality

Server side has seen some changes too. It now listens to client all the time and the client side can turn off the stream (for example, to change resolution) and then continue streaming with new parameters. Sometimes, it may crash if stream ended unexpectedly (if GUI crashed, then server should be booted up again).

It has some limitations though. First of all, resolution cannot be changed while streaming nor recording (hardware limitation). The preview screen should be turned on **before** establishing stream. I really wanted to do the following: enable different resolution for a preview and recording. Sadly, I could not do this. The _resolution_ variable is global for all that is in the PiCamera. I tried working with _resize_ parameter of the _record_start()_ function but it was even worse as resize started dropping the FPS considerably (it does not take smaller resolution from camera, it starts to downgrade bigger resolution). The essential part of this paragraph is that the stream resolution is equal to the camera resolution (drop down menu). The lowest resolution achieves about 27-28 FPS. The highest resolution, actually, works pretty well at about 10 FPS, but it may drop due to, may be, Internet or some other factors. It really hard to tell.

The stream functionality took so long to implement as I was testing possible situations that crash the GUI. For now, if server is not running, client will not be able to connect to it (there is a 2 seconds timeout timer). If server side has ended during stream, which is not advised to do, client side will shut down (sometimes it crashes but mostly not). Some little features like FPS counter after stream ended are included too.

If threading functionality is not welcomed with the 'videobutton', it can easily be disabled - all the necessary steps are in the comments.

Also, I got curious whether it was possible to implement a feature of recording video which is not limited by time. It is done and there is a new button right under the 'record' button. Also, it is possible to stream, record for limited and unlimited time all in parallel to each other. What it means in practice is that we can change temperature on lighting while recording an experiment.

The code is well documented. I commented and cited pretty much every step for my clarity and those who wishes to understand the underlying principles. Yes, I took some inspiration from A. Gibb's book you recommended me 'Building OSHW' and thought that well documented code is a must in our subject area.

This is it for all the main features. I think it would be nice to test them on your side too so we are good to go. [Files](https://github.com/isobianin/Flypi/tree/master/GUI_22)

amchagas, 2020-12-10:
Hi @isobianin !

From this, it looks like you did a very complete job with this implementation! Thanks a lot for this!
I should be able to test things tomorrow, will write more here, so we can go about synching codes and closing issues!

amchagas, 2020-12-22:
Hi Ihor, 
I just checked the new GUI code and it works nicely!
I think this is good for PR an closing this issue

closed 2021-05-27 by amchagas