amchagas/Flypi #37: increase camera resolution

amchagas, 2019-02-03:
Currently the camera/lens combination gives a resolution of 50µm.
To increase the number of applications, it would be great to brings this resolution down to 10-20µm. 

This could possibly be done with:
- Different optics
- Different cameras
- Software to process the data

What is the best cost/performance/easiness of use solution remains to be discovered.

koerli89, 2020-11-04:
Hi,
i have no experience using Raspberry pi oder Flypi but I am thinking about building one. 
My question regarding camera resolution is, if the new HQ Pi Camera module would be compatible with the flypi?
(https://www.raspberrypi.org/products/raspberry-pi-high-quality-camera/?resellerType=home)

amchagas, 2020-11-04:
Hi,
Thanks for reaching out and your interest on the project!

In principle this camera should work as well. You probably would have to redesign the camera holder though as the model being used at the moment and this new one have different footprints. 

Also, most likely the resolution limit for the system is related to the optics, and not the camera being used at the moment (since the camera pixel size is 1.4µmX1.4µm if I am not mistaken). There is a brief discussion about this on the FlyPi paper if you want more details!

I hope this helps!