amchagas/Flypi #38: Improve user manual

amchagas, 2019-02-03:
There is a [PDF](https://github.com/amchagas/Flypi/blob/master/User%20and%20Assembly%20Manual_revised.pdf) describing how to build and use the FlyPi. Unfortunately this document is outdated, since the hardware and software have been updated.

This update has been started and we are now favouring an open online documentation system called [wikifab](wikifab.org). The FlyPi documentation can be [found here](https://wikifab.org/wiki/FlyPi#).

This documentation needs to be extended and finished.

neurologic, 2021-05-07:
I went to the wikifab and see that it shows photos of the latest 3D print parts. But I still cannot find an updated PDF that describes how the parts are supposed to fit together. I can't figure out the necessary orientation of all of the parts relative to each other based on the PDF showing the old parts. Do you have any detailed videos or photos of the new Version 2.0 in action?
Thanks!

amchagas, 2021-05-21:
hi @neurologic, 
unfortunately, not at the moment... this is one on the to do list...  Or might even be something to get an external collaboration of someone setting up a new system for the first time? :P On the other issue you opened up I shared an email for us to figure out an online call... Maybe if I give you an overview on the call you could document your step by step setting up?