amchagas/Flypi #40: Improve z axis micromanipulator stability

amchagas, 2019-06-21:
The current version of the z axis of the micromanipulator consists of an M4 screw and a single locked nut that act as the axis carrier. this makes the axis a bit wobbly. 

A possible solution for this would be to add a second nut at a certain distance from the previous one similar to what is present on the other axis. 

Other improvements to be implemented are:
- Adding a spring (or two) to the axis to decrease backlash
- improving the pockets where the nuts go in.

ODESSARAMBO, 2019-07-02:
Hello. So I have made some progress on this particular issue (#40) and the next one (#41) and submitted proposal to this job on upwork. Here I will dublicate screenshots of design. Can you please tell me whether I need to continue working on this project?
[image]
[image]
[image]

amchagas, 2019-07-02:
Hi @ODESSARAMBO this looks very nice! 

Thanks for the effort in putting it together. 

As you probably saw, that Upwork offer is closed already, since we learned after putting it online that, because of the regulations of the funding agency we received a grant from,  we can only work with freelancers that can emit a bill with a VAT number. 
We created a new job post stating this fact https://www.upwork.com/ab/applicants/1145797064325591040/job-details

I'm sorry about this confusion.   

If you are able to provide us with a bill that has VAT and VAT number, then we can keep discussing things!

best,
Andre

ODESSARAMBO, 2019-07-02:
Thanks for the answer! I will try to get more information about billing with VAT. For now, made a bit more progress on the tightening system

amchagas, 2019-09-11:
Solved by @isobianin contribution via PR https://github.com/amchagas/Flypi/commit/50d048502420fd8981893c8c185ba7404e968846

closed 2019-09-11 by amchagas