amchagas/Flypi #41: improve system to hold samples on manipulator

amchagas, 2019-06-21:
The current systen for holding samples only allows microscope slides to be held and it is based on a tight fit. If a thicker or thinner microscope slide is used, it is a bit troublesome for the user. 

To correct this, a system with a screw lock would be nice. Maybe re-using the printed screw system present in the other parts of the system. Another nice addition would be to have an interchangeable head, where different holders could be placed in, depending on what the user needs in terms of attachement.

Part that holds the microscope slide:
https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/Manipulator_Zaxis.step

Here are the files containing the screw system (as guidance):
https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/NeckTriangle_Thread.stp

https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/NeckTriangle.stp

ODESSARAMBO, 2019-07-04:
Hello. I propose the next system that allows having changeable microscope slides. It functions as C-clamp and consists of 4 parts: screw, frame, top and bottom plates. Frame can be placed on either side and it is fixed in place by m3 screw (however, it can be mounted without them). The hole system uses the interchangeable system from the issue #40. Generally speaking, it is improved design that was included to the second job proposal.
[image]

isobianin, 2019-07-12:
Hello. I see that this issue has not been updated much. Here are step files of parts I mensioned here and in #40. Hope it will help you. Would really appreciate feedback so I know whether you will use or the problem is alredy solved.
Actually have some ideas on your next problems with enclosures. I can do it on voluntary terms but it will take longer time to complete
[holder.zip](https://github.com/amchagas/Flypi/files/3387192/holder.zip)

amchagas, 2019-07-15:
Hi! Thanks for sending this in. This issue hasn't been worked on yet. It looks nice, and I can find some time to print and test it tomorrow or the day after... 

I really appreciate your help!

isobianin, 2019-07-15:
Hi! Okay, got it. Looking forward for the prototyping phase

amchagas, 2019-07-22:
Hey,
I managed to print this out today, and some of the parts are too thin for the printer software to capture and properly slice it, so the parts end up printing "incomplete", for instance the m3 holes on the base do not print.
Also the top and the bottom plates do not slide in the zaxis base as the fit is too tight...

isobianin, 2019-07-22:
Hi. Could you please send me some photos of failed parts? Also would be nice to know the model of your printer. And about that m3 holes. What about the same m3 holes but made in the z-axis detail (#40)? Are they damaged (incomplete) too?

isobianin, 2019-07-22:
And about fit. Are those holes for nuts too tight in width or in height?

isobianin, 2019-07-23:
Hi @amchagas . In order to identify tolerances I propose to use the next set of parts that will help us to achieve our goal. In the archive you will find 5 parts that correspond to the questionable places in issues #40 and #41. Those are holes for nuts (#40) and T-connectors (#41). Moreover, from here and on it would be really good material-wise choice to print questionable parts of the main detail in order to define "printability"
[tolerances.zip](https://github.com/amchagas/Flypi/files/3421985/tolerances.zip)

amchagas, 2019-07-24:
Hi @isobianin 
The printer currently in use is a prusa mk3s https://www.techradar.com/uk/reviews/original-prusa-i3-mk3s

the holes for the nuts are fine (those are M4 nuts and bolts).

The fits that were too narrow only happened for the printed parts that had to slide onto each other. Below are photos of the parts. 

[image]
[image]

I can print the parts in the archive later today. As a reference, adding a 0.1 to 0.2mm tolerance for sliding parts has given us some ok results on previous projects.

isobianin, 2019-07-24:
Hi @amchagas. Thanks for photos and information about printer

isobianin, 2019-07-24:
About tight fit. I assumed that tolerance 0.2 mm would be enough (rule of thumb for most 3d printers) but, as the result shows, some things can only be solved by trial and error

isobianin, 2019-07-25:
Hey. Here is the connector from file Manipulator_Zaxis_base. Increased width of walls where the holes for m3 are
[base_connector.zip](https://github.com/amchagas/Flypi/files/3429769/base_connector.zip)

amchagas, 2019-07-29:
Hi, 
so the conclusion from printing the parts for test was that the smaller versions of both fits (m3 holes and t shpaed) are good. The bigger ones are too large I think.

second and fourth photos are the larger t shaped fit

[image]
[image]
[image]
[image]

isobianin, 2019-07-29:
Hi. I see. Will adjust design asap. I think we could go for tighter fit but we may aswell use this knowledge somewhere else

isobianin, 2019-07-30:
Here are corrected versions of z_axis, clamp base and clamp top. I forgot to mention but you have already printed parts where shaft and holes were changed. I mean, that you will reprint z_axis and base anyway, but the top part of clamp (where small cylindric detail is) may have to be reprinted. But I think that printed part should fit too, but it may be a little bit loose.
Also, I increased width on walls that connects base with pillar so now it should be printable.
[corrected.zip](https://github.com/amchagas/Flypi/files/3445621/corrected.zip)

isobianin, 2019-07-30:
I have made all necessary changes. Also I have increased width of walls that connect base with pillar. I forgot to mention but the hole T-system had slight dimension rework. You are going to reprint z_axis and base anyway but the top (with little cylindric detail) may have to be reprinted too. But I suppose that the current part will fit due to the tolerances but it may be a little loose 
[corrected.zip](https://github.com/amchagas/Flypi/files/3445673/corrected.zip)

amchagas, 2019-07-31:
Here are the printed parts: 
[image]
[image]
[image]

The fits seem alright for now, but the plates do not have enough space in between them. I mean the base of the zaxis has the negative t-shape, but it is not long enpough so that the plates are far a part enough, as can be seen in photo1. 

also the threaded part could be a bit longer (longer thread), and it would be great if the handles could start stright at the top (would make it easier to print it)

isobianin, 2019-07-31:
Hi. I see. How much longer do want the thread to be? About handles - definetly my fault, I thought about this when it had already been printed. And about the length of t-hole in z-axis, somehow it just slipped my mind. How thick will (aprox) the thickest microscope glass be? I will increase depth based on that.
And another question. So does the core tightening system work? I mean, does it tighten slides inbetween and does the ball on screw and its connector need any tolerance rework? Just to be sure so there will be less material loss.

isobianin, 2019-07-31:
Increased depth to 13 mm (pic.1)
[image]
Pic. 1 - New depth
Patched bolt handle and added +5mm to thread (before it was 15mm)

isobianin, 2019-07-31:
[changed.zip](https://github.com/amchagas/Flypi/files/3452827/changed.zip)

isobianin, 2019-07-31:
So now it can hold slides up to 3 mm in thickness without moving out of bounds

amchagas, 2019-08-09:
Also printed these guys and it is looking good. however the top plate doesn't slide down when the screw is turned. What ends up happening is that the top plate bends downwards when enough turns are made on the screw. In the last image it looks like this is causing strain on the other joints. 

I wonder if a system where both plates were fixed (with 4 mm distance or so) and a printed screw could go trhought the top plate would be enough? a little bit like the clamping system used for the other movable parts?

[image]
[image]
[image]
[image]

isobianin, 2019-08-09:
Ok, I will send here some sketches that illustrates your proposed design as I understand it. Also I was thinking about this issue after completion of the last one #45. I will attach sketch of my idea too

isobianin, 2019-08-09:
So here is the finished simple system that I actually proposed as the second variant during the time we started this project. Also, reduced material usage by creating gaps in details. Picture 1 has annotations in it that explains assembly.
[image]
Pic. 1 - Simple M3 thread system
Attaching step files so you can test it right away
[holder m3.zip](https://github.com/amchagas/Flypi/files/3486518/holder.m3.zip)

isobianin, 2019-08-09:
How I see your technical proposal (pic. 1)
[image]
Pic. 1 - Technical proposal

amchagas, 2019-08-09:
So the drawing is what I'm thinking about, yes. 
I can print the other holder as well to compare

isobianin, 2019-08-09:
So here I made your variant (pic. 1 and pic.2) and combined version (pic. 3). Fixed position variant has 4 mm clearence.
[image]
Pic. 1 - Variant with fixed position
[image]
Pic. 2 - Slice of model
[image]
Pic. 3 - Top plate with threaded hole and wings for M3 screws
And bolt for tightening slides (pic.4). Has 20 mm thread legth
[image]
Pic. 4 - Bolt (handle from #45)
[holder fixed.zip](https://github.com/amchagas/Flypi/files/3488066/holder.fixed.zip)

amchagas, 2019-08-14:
Hi, 

this looks quite nice too!
I'll try to pop itin the printer today!

isobianin, 2019-08-14:
Hello. Ok, looking forward to seeing it. Just to be sure (because this issue has a lot of things) there is a zip named holder m3.zip and holder fixed.zip. Totally there are 3 variants of the same thing but the last archive (holder fixed.zip) is like definiteve edition where all things can be tested:
- nut holding
- applying pressure via bolt
- combined version with both methods

amchagas, 2019-08-14:
cool, thanks!

amchagas, 2019-08-14:
also, I had already printed the other clamping solution, but the nut pockets were too big.
[image]

isobianin, 2019-08-14:
Yeah, I see it. I suppose it still can function as a test object. Those tolerances are taken from holes that we were testing in order to find the right sizing. I will look up my files, maybe something went wrong or we can make holes smaller

isobianin, 2019-08-14:
Those are nuts for m4 thread, sorry. Somehow, mixed it all up. So holes are probably too tight for m4 screws. I will make them proper diameter. I think, that it is better to stick with m4 nuts because they are a bit bigger than m3 and so less likely to have issues with size distribution (mainly 3d printer prints sizes differently every time and smaller we go, more precise equipment is needed)

isobianin, 2019-08-14:
Basically, it is previous archive but with m4 holes
[holder m4.zip](https://github.com/amchagas/Flypi/files/3500910/holder.m4.zip)

amchagas, 2019-08-15:
I think we have a winner here. Below are the photos of the fixed system with only one printed screw. It holds well, even if the part is quite large and "heavy"

[image]
[image]

If you are ok with this, I would use this design and call this issue and the other one related to Z axis stability closed?

isobianin, 2019-08-15:
Good job! Sometimes less is more. Just the right thing for current issue

amchagas, 2019-08-16:
cool, do you want to make a pull request with this final version so that your name can be added to the contributors list and we can close this and the other related issue?

isobianin, 2019-08-16:
Hi. So how exactly can it be done? I just cannot find the button

amchagas, 2019-08-16:
I see you created a fork of this project, so the best way to create a pull request would be to follow these guidelines: https://help.github.com/en/articles/creating-a-pull-request-from-a-fork

isobianin, 2019-08-19:
Hi @amchagas ! So I have uploaded all files that we have designed together. As I understand this, I will create pull request when we have design in #43, #44 and #45 settled. I created folder with my surname in /STEP files folder to keep things organized.
P.s. Aforementioned issues have been updated. Just waiting for approval

amchagas, 2019-08-21:
Hey @isobianin, yes, you are completely right! 
I'm printing the updated versions and can let you know when I'm done.

isobianin, 2019-08-21:
Hi! Okay, I get it with pull requests, thanks.
Looking forward to seeing print results

isobianin, 2019-08-26:
Hi! What is the situation with the details? Is everything alright?

amchagas, 2019-08-26:
Hi!
so the base is printing nicely. The pcb case cover and the system to hold petri dishes I documented here https://www.dropbox.com/sh/418cd881mosbbwr/AABIq8Ff3xSaW14b0SKrVxpJa?dl=0

isobianin, 2019-08-26:
Okay, I see. So there is a small tolerance problem with pcb case. Strange, was it there when you have printed it last time? I will make adjustments in model. I hope that it is still usable or is it so bad that you cannot open it?
Speaking about holder, I see that there is a backlash. Is it present in other places where the same srcew system is used? And what about other parts of the holder. Have you tried assembling it? Just want to know whether the whole system is working.
Please, correct me if I have missed something.
P.s. Is there a way to contact you via email? I would like to discuss same matters if it is possible

amchagas, 2019-08-26:
Hi, 
This is still usable! don't worry. yeah, I don't know if we didn't notice that little thing because before it was getting stuck in the inner thinner part. 

I don't see that much backlash on the other parts the screw system is being used, but then again the screws are shorter, and have other purpose. The other thing I can comment is that the rectangular slides are ok, but a bit on the thin/"too delicate" side, also they tend to slide out of the rails a bit (maybe a "V" system would do better?). As far as I remember the other parts were fine, and they fit nicely when assembled as a whole. 

you can always send me a message @ andremaia.chagas[at]gmail.com! but keep in mind this potentially keeps other people out of this discussion thread. 

Once again, thanks for all the help with this project!! really appreciated! 

You can send me an emai

amchagas, 2019-09-11:
Solved by @isobianin contribution via PR https://github.com/amchagas/Flypi/commit/50d048502420fd8981893c8c185ba7404e968846

closed 2019-09-11 by amchagas