amchagas/Flypi #43: acessing cables without unmounting the boxes

amchagas, 2019-06-21:
In order to plug and unplug modules, users need to disassemble at least the PCB box. 
What we want is to create a click on or flipping system where the modules can be plugged and unplugged in an easier manner. 

Here are the files for the PCB box, where the modules are plugged in and out:

https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/PCBCase_Top.stp

https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/PCBCase_Bot.stp

isobianin, 2019-07-20:
Does the highlighted hole serve functional purpose? What exactly it does?
[image]

isobianin, 2019-07-20:
I have in mind a pair of hinges that will be placed either there or where the hooks currently are. I thought that this side of enclosure is less likely to be loaded with ports but I do not know the used PCB layout

amchagas, 2019-07-22:
Hi, sorry it took me a moment to take a photo of the case with the cables and a part of the pcb.

[image]
[image]

isobianin, 2019-07-22:
Thanks for the photo! It is quite informative! However, I still have questions. Is the opposite side of the box (pcb) loaded with cables or ports too? I don’t see it in this photos, but structure of the enclosure have holes for ports there.
It is important for the hinge system I want to implement

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

isobianin, 2019-07-22:
Yeah, I see that both sides are loaded

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

isobianin, 2019-07-29:
Hello @amchagas . I am almost finished with the new design. For now a test for hinges is needed 
[hinge_test.zip](https://github.com/amchagas/Flypi/files/3442756/hinge_test.zip)

amchagas, 2019-07-29:
hi @isobianin, great, thanks! can print these tomorrow

isobianin, 2019-07-29:
So this is what I came up with. The first picture demonstrates hinges placement
[image]
The second one illustrates clicking system from this same enclosure with stifeners (as in #42)
[image]
In the third picture you can see how the case looks being open
[image]
Attaching step files in case printing phase goes well and hinges fit.
[PCB_mod.zip](https://github.com/amchagas/Flypi/files/3443610/PCB_mod.zip)

amchagas, 2019-08-09:
Hey!
so I managed to print this, and there is one main issue, namely that the tabs in the bottom part are too thin and they broke right off in the first opening of the system... I wonder if making them thicker would be too much trouble?
[image]

isobianin, 2019-08-09:
Hi. Okay, I will correct strength of the detail. Shame that this occurred on the printed parts

isobianin, 2019-08-10:
Increased thickness of wall towards hull (pic. 1)
[image]
Pic. 1 Thickness
[PCBCase_Bot_mod_v2.zip](https://github.com/amchagas/Flypi/files/3488750/PCBCase_Bot_mod_v2.zip)

isobianin, 2019-08-11:
Created another variant of hinges that are more stable, however, they take a bit more place in the interior (pic.1-pic.4)
[image]
Pic. 1 - Bottom
[image]
Pic. 2 - Top (outside view)
[image]
Pic. 3 - Top (inside view)
[image]
Pic. 4 - Assembled
[image]
Pic. 5 - Opened
[PCB_case_alternate.zip](https://github.com/amchagas/Flypi/files/3490288/PCB_case_alternate.zip)

amchagas, 2019-08-15:
Hi! so here is the new version printed. 
It looks quite nice and the hinges do not break :) 

[image]
[image]

amchagas, 2019-08-15:
I think we are ready to call this one closed?

isobianin, 2019-08-15:
Here is patched version (pic. 1)
[image]
Pic. 1 - Short version
Actually, having step file with pcb layout would have been nice, but are almost finished with this one
[pcb_case_short.zip](https://github.com/amchagas/Flypi/files/3505270/pcb_case_short.zip)

isobianin, 2019-08-28:
Adjusted tolerances based on the latest (26.08.2019) prototype
[PCBCase_Bot_v2.zip](https://github.com/amchagas/Flypi/files/3550609/PCBCase_Bot_v2.zip)

amchagas, 2019-09-11:
Solved by @isobianin contribution via PR https://github.com/amchagas/Flypi/commit/50d048502420fd8981893c8c185ba7404e968846

closed 2019-09-11 by amchagas