amchagas/Flypi #44: base optimizations

amchagas, 2019-06-21:
The base could use some optimizations:

1 - Currently it takes quite a while to print, it would be useful to hollow out some parts so that it prints faster. It should be kept in mind that hollow parts require some sort of support/bridge to print properly.

2 - on the top side of the base, where the peltier cables go in, one of the sides is higher then the other. we want to fix that by making both sides as low as the current lowest side and to make the slits curved and not squared as they are currently (this will lead to better finishing when it is printed)

amchagas, 2019-06-21:
https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/Base.stp

isobianin, 2019-07-21:
Want

isobianin, 2019-07-21:
Here I have multiple questions:
1. Is this an orientation for the printing?
[image]
2. What are the functional purpose of all holes and how does this part connects to the others? Some triangular holes are blocked (picture below). Need to know more because I want to create holes as shown in the picture.
[image]
3. Have a problem with the second point of this issue. Do the circled points show the slits for peltier cables (picture below)? At least, I think so. In this case, there is now difference in height between slits (pink lines in the picture)
[image]

amchagas, 2019-07-22:
hi @isobianin ,
to answer your questions:
1 yes, the base is printed upside down, as you show in 1.

2. the triangular holes are used as fits for the necks (https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STLs/Neck_L.stl). The pieces "blocking" the necks are not actually blocking this fit, and they are there to hold a fan in place (placed from below to be sitting under the peltier element).

3 yes, this part of the issue was resolved already.. sorry for not updating the description.

isobianin, 2019-07-22:
Hi @amchagas , thanks for information. But the slits are still need to be curved, right?

amchagas, 2019-07-22:
You mean so that the cables fit in them? I think it is not that pressing, since squared slits are working fine too..

isobianin, 2019-07-22:
Okay, leaving it squared

isobianin, 2019-07-31:
Hi @amchagas . Here is an update on this issue. Picture one shows amount of material before.
[image]
Pic. 1 - Slice of model (before)
The next one - after rework.
[image]
Pic. 2 - Slice of model (after)
[image]
Pic. 3 - Light version of the base
For example, numeracly it can be shown as following:
V = 62330.061064 mm3 (before)
V = 22574.223411 mm3 (after)
Almost 3 times lighter. 
Attaching file to this comment and waiting for your approval of this design
[Base_light.zip](https://github.com/amchagas/Flypi/files/3450480/Base_light.zip)

amchagas, 2019-08-14:
Hi, 
So I managed to print this and indeed it cut the printing time in half!
There is one issue though that the most inner wall is too thin and breaks easily... would it be hard to increase the wall thickness a bit, or add reinforces along the wall? (as long as it doesn't cover the triangular holes?
[image]
)

isobianin, 2019-08-14:
Okay, I will make necessary adjustments

isobianin, 2019-08-14:
I have looked up my model. So is there a problem in all other walls too? Because they are all the same and have 1.5 mm thickness

amchagas, 2019-08-16:
I think the problem is mostly with the inner ones... maybe the bridges and other elements closer to the outter walls somehow make them stiffer?
Or maybe there is an issue with the slicing software, but then this is something that is not easy to change from our side...

isobianin, 2019-08-16:
Hey. Increased thickness of inner walls by 1 mm. (pic. 1, pic. 2)
[image]
Pic. 1 Reinforced model
[image]
Pic. 2 Interior
[base_light_reinforced.zip](https://github.com/amchagas/Flypi/files/3509923/base_light_reinforced.zip)

amchagas, 2019-09-11:
Solved by @isobianin contribution via PR https://github.com/amchagas/Flypi/commit/50d048502420fd8981893c8c185ba7404e968846

closed 2019-09-11 by amchagas