amchagas/Flypi #45: Holder for petri dishes

amchagas, 2019-06-21:
The current petri dish holders are fixed to two different sizes. It would be nice to have a clamping system that is able to handle dishes and other containers with different sizes. 
It can use the thread/screw system already implemented on the system.

here are the files for the petri dish holders:
https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/PetriDish_56mm.STEP

https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/PetriDish_36mm.STEP

and the thread/screw system:
https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/NeckTriangle_Thread.stp

https://github.com/amchagas/Flypi/blob/master/3D_print_files/v2.0/STEPs/NeckTriangle.stp

isobianin, 2019-08-06:
Hi @amchagas ! Are there any news about previous issues? I have been working on this since then and have some things to show.
There are two variants of the vise that supposed to replace old holding system. First variant uses stift to hold the movable bolt in place and consists of movable and immovable jaw placed on rails, bolt with the thread (the one you already have in your project) and dubels to hold cylindric and other forms with diameter up to 70 mm (pic. 1).
[image]
Pic. 1 - Vise with stift
The slice of this model is shown below (pic. 2).
[image]
Pic. 2 - Slice of the vise with stift
The problem here that bolt is held by only stift. So there is another variant that has longer bolt so the structure can be disassembled (pic. 3).
[image]
Pic. 3 - Vise with prolonged bolt
The next slice illustrates how bolt is stopped.
[image]
Pic. 4 - Slice of the vise with prolonged bolt
Movable jaw looks like this (pic.5)
[image]
Pic. 5 - Movable jaw
So this variant is arguably more durable and secure but may come undone when used in its maximum holding position (diameter more than 70 mm) (pic. 6)
[image]
Pic. 6 - Disassembled vise with prolonged bolt
For reference, I have place 56 mm holder in this system (pic. 7)
[image]
Pic. 7 - Reference assembly

isobianin, 2019-08-06:
[vise.zip](https://github.com/amchagas/Flypi/files/3472472/vise.zip)
There is note in this archive in order to assemble chosen vise correcrtly without printing unnecessary parts

amchagas, 2019-08-07:
Hi @isobianin !
this looks quite nice! Thanks for taking the time to develop it :) I had some time problems last week and the printer also acted out... So I'm still printing the other parts and can pop this in when those are done.

isobianin, 2019-08-07:
Hi! Ok, looking forward to seeing the results of the latest prototypes

amchagas, 2019-08-14:
I printed this one as well, and there was an issue with the sliding T parts... sorry!

Namely they don't print so well without support, I wonder if having them as V shaped would be too complicated? this would facilitate printing and possibly making sliding easier...

isobianin, 2019-08-14:
Got it

isobianin, 2019-08-14:
Changed rail profile so now it is just simple rectanle (the safest of all options) (pic. 1)
[image]
Pic. 1 - New profile
[vise_v2.zip](https://github.com/amchagas/Flypi/files/3501053/vise_v2.zip)

isobianin, 2019-08-28:
Deacreased tolerance in the immovable jaw nut (pic. 1 and pic. 2). 
[image]
Pic. 1 - Before
[image]
Pic. 2 - After
File to test this tolerance here:
[nut_test.zip](https://github.com/amchagas/Flypi/files/3550808/nut_test.zip)

isobianin, 2019-08-29:
V-shape rails added (pic. 1)
[image]
Pic. 1 - V-shape rail
There was a problem with the rails that I have been thinking. What bothered me is that they just hanged in space. I added support in the movable jaw, but now jaws have some offset (pic. 2). 
[image]
Pic. 2 - Sizing
Files to print if previous tolerance test goes well
[vise v3.zip](https://github.com/amchagas/Flypi/files/3554162/vise.v3.zip)

isobianin, 2019-08-29:
I have designed another variant of the vise. In this system rails are now printed separately and then mounted in the immovable jaw (pic. 1 and pic. 2)
[image]
Pic. 1 - Front
[image]
Pic. 2 - Back
Rails have slots for screwdriver just in case. This vise does not have any offsets but it is now 5 mm higher
[vise_screws.zip](https://github.com/amchagas/Flypi/files/3556639/vise_screws.zip)

isobianin, 2019-09-03:
Hi @amchagas ! What about models? Is everything good?
Also, I have a question. If our designes are used by other researchers via 3d printing, is there a necessity of creating assembly instructions? I mean, just a set of pictures with parts' names and maybe some sequences of actions. Basically, pictures that I send when design is finished but with the latest versions of each part

amchagas, 2019-09-04:
Hi @isobianin things look good!
the round screwable rails are definetly a nice way to do things! below are some photos. I think the only commentary I have is that the lock system for the screw is a bit too thin for 3D printing, I broke one trying to plug it in and out...  (see first image to know which part I mean)
Creating instructions for the system would be nice add on, if it is really just the pictures you sent.. otherwise I wouldn't want to you to have the extra trouble.
[image]
[image]

amchagas, 2019-09-04:
I think this one can be closed if you don't have more time to work on it, as this last iteration is quite good!

isobianin, 2019-09-04:
I am happy that this device works as intended. Thank you for appreciating the time I spend. I just want to make sure that every system that was designed or modified during this period is done as best as I can do it. I will make instructions soon just to wrap everything up nicely. For me this kind of work is just too interesting
A new bolt needs to be tested (pic. 1). The trick here is to identify whether it will print such overhanging structure (I do not see a problem here because we have already done it with smaller diametres but who knows what may happen)
[image]
Pic. 1 - Bolt lock system
It is just a regular bolt that is missing a handle and shortened to the minimum. Movable jaw included just to test tolerances (pic. 2)
[image]
Pic. 2 - Movable jaw part for test
[lock_test.zip](https://github.com/amchagas/Flypi/files/3575685/lock_test.zip)
Archive if test is good
[vise_screws_v2.zip](https://github.com/amchagas/Flypi/files/3575703/vise_screws_v2.zip)

isobianin, 2019-09-05:
Structure (pic. 1) and instructions for assembly (pic. 2). If something is wrong or written unclearly, please tell and I will correct it. Small changes will not take much time. Also, pictures may have some mistakes in their texts so please tell me if you find any. 
[image]
Pic. 1 Structure
[image]
Pic. 2 Assembly
P.s. I see that this device is the only one that actually needs some kind of clarification (microscope slides holder has only two parts, pcb enclosures too). But I may be wrong

amchagas, 2019-09-09:
Hi @isobianin , 
the instructions look great! thanks for putting them together. 

The test part printed also printed well! 

I think this closes all the issues? or did we still have something to check/change?
If they are all done, I would ask you to make a pull request with the latest files and we close the issues once the PR is made.

isobianin, 2019-09-09:
Hi! Okay, good job! I will double check files just in case, look into issues one more time and than make a pull request. It will be a matter of one or two days

amchagas, 2019-09-11:
Solved by @isobianin contribution via PR https://github.com/amchagas/Flypi/commit/50d048502420fd8981893c8c185ba7404e968846

closed 2019-09-11 by amchagas