amchagas/Flypi #46: Servo motor & the 2nd version of 3D printed LED matrix holder

LabuserTu, 2019-06-25:
Hi, first thanks for the detailed instructions for such a cool microscope! We´ve managed to build one, using the PCB version 0.6  and the 2nd version of 3D-printed parts (some photos attached). It works as expected - a manual, together with some forum discussions were helpful enough to make the basic system running. Still, we have some extra questions and would be grateful for your opinion. The prometheus science forum seems like a better place to open this issue but I was somehow unable to register neither normally nor via Twitter. 

We´ve implemented the automated focusing of the RaspPi Camera but it seems, with the printed cogwheels provided, that it can´t really do it´s job? Servo´s limited 180 degrees of rotation covers only a small range of the lens´ thread which enables focus adjustment. Since the cogwheels are quite tightly attached to each other, it is also not possible to roughly adjust the focus manually, at the same time the cogwheels are attached. Have you also encountered this issue or is there a better way to construct the automatic focusing? We´re using Reely Analog Servo S-0009 MG. Besides increasing the cogwheel size, we were thinking a bi-directional DC motor could help us with the issues. Maybe something similar as it is described here:  https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/overview ? Or is there a specific reason the servo motors are used? Any suggestion would be appreciated. 

Also, as it seems servo motor cogwheels and the LED matrix holder STL files are missing in the second version. Maxime kindly sent the cogwheel STL files. As for the holder, I´ve read that you haven´t update it due to lack of public interest. Just in case, I wanted to ask, wheather it has been created in the meantime?

Lastly, it´s not really crucial but we have a problem with the zap function in the LED1 module. It works well for the LED Ring but nothing happens when the zap button is pressed to control the timing of  high-power UV LED. I´ve seen that the arduino sketch (flypi_development.ino) in the LED1 section doesn´t include similar zap parts as in the LED Ring section - is this why it doesn´t work and should still be added?

Thank you very much!

[image]
[image]
[image]
[image]
[image]

amchagas, 2019-07-01:
Hello, 
great that you are building one of these :)

Yes, it would be great to have this conversation in the forum so that we keep all issues related to usability and problems during build in one place. I took a look at the forum and it seems a recent update on the database broke the signing up system. Could you try it again and let me know how it goes? www.forum.prometheus-science.com
Thanks!

LabuserTu, 2019-07-03:
Hi.

thanks for the reply.
It worked, I registered and created a new discussion there.

Best,
Nina

amchagas, 2019-07-04:
This has moved to a thread in the FlyPi user forum: https://www.forum.prometheus-science.com/index.php?p=/discussion/615/servo-motor-the-2nd-version-of-3d-printed-led-matrix-holder#latest

closed 2019-07-04 by amchagas

amchagas, 2019-10-23:
Hi @LabuserTu please take a look at this thread: https://github.com/amchagas/Flypi/issues/47
@isobianin has worked on a holder for the matrix!