amchagas/Flypi #47: Create a matrix holder for version 2 of the printed parts

amchagas, 2019-07-04:
It seems more people than we thought are using the LED matrix that was published with the paper. Since there wasn't much use from our side, we didn't port the holder of version 1 to version 2 of the printed parts. 

We want to create a holder for it. The matrix is this one: https://www.adafruit.com/product/870

isobianin, 2019-09-17:
Hello @amchagas ! It is this and #34 issues I have referenced in the email. Actually, I am writing this because I have not yet recieved any form of notification from you that my letter had been read or at least seen. More than sure that it is either in "spam" folder or it is lost somewhere in other emails. The subject is "Summary of the work done and further cooperation". 
Regarding this issue, how holder is mounted on the microscope. Or is it a standalone object? What are files for previous model?

amchagas, 2019-09-17:
Hi @isobianin ! 
Just replied to your email. Sorry for the delay!

The matrix is the one mentioned on my comment above here, and the holder is this one https://github.com/amchagas/Flypi/blob/master/3D_print_files/v1.0/individual_stls/matrix.stl (there is also an STEP version on the same folder).

The holder should work similar to the camera or ring holder of the current printed version, so something that can slide up and down the "necks" and that can be locked in place with a printed screw.

isobianin, 2019-09-18:
Hi! Okay, I will look into these files and see what I can do. Possibly, there will be more questions to come. 
Also, what about heat dissipation? Is everything alright with it? This question bothers me because of 2 things:
1) I have been browsing through files one day and have found file named something like "heat dissipator for LED". Probably, there was or there is a problem with that. And this ties to the second thing;
2) Surprisingly, in my diploma project I am developing methodic for designing LED radiators with a help of open source software. At least, we have a chance to simulate this matrix if such necessity exists.

isobianin, 2019-09-19:
Hi @amchagas. So here are more questions (fig. 1).
[image]
Fig. 1 - Questions
1) What is the purpose of this hole (red circle)?
2) How exactly is matrix held in place? I thought that this holder would have some kind of sufrace it can lay on. At least I wanted to implement it in a new version because I see that this design is quite unstable for LED. Or can it be that the matrix is simply placed on top of rectangular hole?
3) There is a certain distance between the centre of rectangular hole and the centre of the neck. Is it critical or it may differ in the new version? I say that it may differ up to 10mm

isobianin, 2019-09-20:
I have designed two variants (fig. 1, 2)
[image]
Fig. 1 - LED holder
[image]
Fig. 2 - Alternative version (file name LED_holder_rectangle)
During the design I had some troubles with dimensions that are mensioned on the developer website so I took all the necessary data from PCB layout files. Assembly check (fig. 3)
[image]
Fig. 3 - Checking assembly
[LED holders.zip](https://github.com/amchagas/Flypi/files/3634483/LED.holders.zip)

amchagas, 2019-09-23:
Hi @isobianin !
Some answers:
The heat dissipators are only needed for a high powered led we are using for fluorescence imaging, but this is already covered with this [part/mount]() where the heat dissipator is glued to the led and the whole thing is fitted into the printed part. 

This Matrix doesn't heat up enough to be a concern.

1 - On fig1, it is indeed that hole is for cables. That can be better implemented though, as in the current state, one will need to unplug the cables from the board if the holder needs to be exchanged (in case it breaks, etc). 

2 - the distance of the center of the matrix to the center of the neck hole is based on two things: a) on the center of the camera when placed in its holder (LED matrix center and camera lens should be aligned), b) the matrix itself is connected to something adafruit calls a "backpack" which is a pcb board that is extending on at least one axis beyond the border of the matrix itself.

3 - The matrix is held in place by using the fact that the backpack behind the matrix is longer than the matrix itself, and a little bit of a tight fit around the matrix, so that the whole assembly can be turned in any position. 

I can print these out today and see how they fit with the matrix.

Thanks for putting this together!!!

isobianin, 2019-09-23:
Hi! Thanks for clarification. Luckily, I have guessed most of the answers.  Hope that prototypes will be alright.

isobianin, 2019-09-27:
Hi @amchagas ! Is there any update on this issue? And what about #34? I have left my commentary and thoughts there too

amchagas, 2019-09-30:
Hi @isobianin ,
sorry for the delay again. 
seems there was a problem with the scale of the matrix holder (see below
[image]
)

isobianin, 2019-10-01:
Hi Andre! That is odd. I can make necessary adjustments very quickly. However, you are showing me different matrix than the one mentioned in issue. Your post mentions a holder for https://www.adafruit.com/product/870 If we go to the website, we see the following (fig. 1)
[image]
Fig. 1 Adafruit Mini 8x8 LED Matrix w/I2C Backpack
Backpack dimensions are mentioned later as 20mm x 28mm x 4mm / 0.8" x 1.1" x 0.2"
Here it is as the footprint (fig. 2)
[image]
Fig. 2 - Footprint
So I made this holder for this matrix. 
I suppose, that matrix in your photo is this one https://www.adafruit.com/product/902 (due to the rectangular led-s)? (fig. 3)
[image]
Fig. 3 - New matrix
And its footprint is this? (fig. 4)
[image]
Fig. 4 - New footprint
So should I adjust holder (I will save both variants of previous matrix and the new one)? Is the new information about matrix accurate?

isobianin, 2019-10-01:
So I have made adjustments with an assumption that we are using this https://www.adafruit.com/product/902 bi-color LED.
Checked assembly in CAD (fig. 1).
[image]
Fig. 1 - Assembly check
[LED_holder_bi_color.zip](https://github.com/amchagas/Flypi/files/3677651/LED_holder_bi_color.zip)

amchagas, 2019-10-02:
Hi @isobianin ,
so sorry, I obviously linked the wrong board!
Thanks for updating the part, I'll print it here in a couple of hours and let you know

isobianin, 2019-10-02:
Hi, @amchagas ! Okay, looking forward to seeing the results

amchagas, 2019-10-02:
The matrix fits and the screw holes are aligned, so one can certainly fix it to the plastic holder that way. There is a small gap between the matrix and the printed part (last photo)
[image]
[image]
[image]
[image]
[image]

isobianin, 2019-10-02:
Here is an interesting thing. I have corrected dimensions and made new design for monochromatic LED. The thing is that the length between centers of mounting holes is shared between bi- and mono- LED. Thus, it makes already printed holder universal. However, I do not know why there is a gap, especially with given circumstances. Could it be that this particular matrix has some kind of offset?

Here are files for monochromatic LED but aforementioned dimensions render these files almost useless. Please, tell me whether I should download them to repository or the previous pack of holders is enough.
[LED_holder_mono_color.zip](https://github.com/amchagas/Flypi/files/3682974/LED_holder_mono_color.zip)

amchagas, 2019-10-03:
Hi! 
I'm really sorry about not noticing the difference between the monochrome and the bicolor versions. I thought the backpack would be the same for them. 

the issue could also be that the Matrix I have here is at least 3 years old, maybe they've updated components.

I think I'm happy with the pack already made, since it can be used for both matrices? So if you are also happy with it, send them to the repo. Other than that I can print the last version you made and see what it looks like.

isobianin, 2019-10-03:
Hello @amchagas !
I am glad that we put effort into finding out the cause of disinformation. Now I have one more thing to add. I am sorry but today is the day when I have finally understood what problem with cables you have referenced in one of your previous posts. This means that I prepared one more version of holder. It is a dead simple concept but I think it will be enough to solve this subissue efficiently (fig. 1).
[image]
Fig. 1 - Cave design
This new cave that is 13 mm long is supposed to be used as an excess room. This room allows LED to be taken out (measured with the highest matrix). Seems that this design is quite good but cables may be touching PCB (backpack). I mean I am being picky here but, still, it is what we are going to end up with. It looks like these cables will not interfere with LED ability to work especially with given PCB layout.
LED needs to be uplifted and rotated in order to be taken out (fig. 2)
[image]
Fig. 2 - LED position for disassembly
Yes, the cave may be mirrored to the other side of holder. But I want this design to remain somewhat rigid and do not want to weak support structures more than they need to be.
[LED_holder_universal.zip](https://github.com/amchagas/Flypi/files/3686673/LED_holder_universal.zip)

amchagas, 2019-10-07:
Hi @isobianin 
thanks for preparing these. 
I'm printing them out now, will let you know how they turn out.

isobianin, 2019-10-07:
Hi! Okay, looking forward to it

amchagas, 2019-10-08:
Hey @isobianin this looks pretty nice! I think we can close this one too.

thanks a lot!

[image]
[image]

amchagas, 2019-10-08:
There is some "brim" from the print that I forgot to remove..

isobianin, 2019-10-08:
Hi @amchagas ! Indeed, things seem to be alright. Have you tested whether it can be disassembled without getting cables out from LED (it is still bothers me)?

closed 2019-10-23 by amchagas