amchagas/Flypi #48: GUI improvement - create classes

amchagas, 2019-07-04:
We've been working on a version for the GUI using pyqt5. this is on the serialupdate branch. 
the GUI is already functional but the files organization is a bit chaotic, there are two files, one describing the entire GUI and one having all callbacks.
We want to create a class for each part of the GUI so things are better readable/organized and easier to add and remove as they will be more modular.

amchagas, 2020-10-01:
@isobianin This is related to issue #59 But again, let's chat more before tackling it

isobianin, 2020-10-14:
Hi @amchagas !

While working on the #59, I had to make some minor and moderate changes to the structure of .ui file due to the fact that the original structure was not compatible with the changes needed. Particularly, widgets were not set in proper layouts and thus they acted a bit weird. Now it is a structured modular .ui file. (side note: I used Qt Designer as I was not able to compile C++ files with Qt Creator due to some internal issues). 

If we are talking about main.py, I think that it is all alright as it is now.

The question here will mirror one in #59, is there a particular point to start? Do we really need to create classes if things are already somewhat organized in .ui?

amchagas, 2020-11-18:
Hi @isobianin ! 
If things are organized on the .ui file, then I think this is fine. I think my original idea was to create classes so that we could re-use bits of the .ui code for other GUI and so on. But then again we are both using Qt Designer, which spits out an .ui file, correct?
If that is the case, I would just close this issue.

isobianin, 2020-11-19:
Hi @amchagas !
I understand where are you coming from. I also believe that this is partially done by Qt Designer in the current state, because a .ui file saves everything via classes. Of course, it could be optimized even more by "hard" coding everything. However, it is highly likely that this will lead to a situation where only few people can really continue doing something with the existing GUI and the time needed to understand the code without Qt Designer is going to increase dramatically

closed 2020-11-19 by amchagas