amchagas/Flypi #59: scroll bar

AndresGarelli, 2019-11-13:
Hi,

is it possible to add a scrollbar to the main window? Depending on the screen resolution, the main windows sometimes does not fit in the screen and some buttons are not accesible. I tried increasing the resolution in the Raspberry pi, but for an unknown reason, VNC does not show the desktop if it put in the highest possible resolution
thanks

amchagas, 2019-11-17:
Hi @AndresGarelli ,
thanks for reaching out and we are glad to hear people are using the Flypi,
I've started looking into this and should have a bit more time during the next days... Hopefully it is an easy fix.

amchagas, 2020-10-01:
@isobianin This is one of the issues I mentioned earlier. But let's talk about it once more before deciding on how to tackle it.

isobianin, 2020-10-01:
Hi @amchagas , I see it. My main question is whether there is a distinctive point from which we can start working. What I mean is that I am not that qualified yet in this area to see a clear solution or even a plan how to tackle it. I believe, as you mentioned in the previous reply here, it should not be that difficult. I know that for web pages there are means to automatically scale web page interface, or GUI in our case, so at least the general solution algorithm should be related to that in some way or another. If there are already some notes, I would be happy to look into them

amchagas, 2020-10-02:
Currently the GUI is written in PyQT5, using QTcreator. There area a couple of threads on Forums showing how to implement something similar such as [this](https://www.learnpyqt.com/courses/adanced-ui-features/qscrollarea/) and [This] (https://www.programcreek.com/python/example/108074/PyQt5.QtWidgets.QScrollBar)

Another solution which I think is a bit of an overkill (at least for me without much experience with Html, css, JS etc) would be to redo the GUI code using technologies that make use of the browser. [Python dash](https://plotly.com/dash/) I think would be a good candidate for this solution since it has a large community backing it, and it keeps the code in Python/.

I think it would be easier to simply add scrollbars to the current GUI.

isobianin, 2020-10-05:
Hi @amchagas ! This seems like a good point to start. And I agree with you that we should incrementally upgrade it to the desired variant. But it would be wise to begin it with the srcollbar.

I will look into it and post new information here

isobianin, 2020-10-10:
Hi @amchagas !
I have been looking into pyqt5 since we last discussed this problem. To begin, I have this window
[image]
Fig. 1 - Main window
I guess that I encountered some compatibility issues with widgets being a bit cut off. But the question here is whether a scroll bar should be added to all submenus(?) and should it be both vertical and horizontal? At least, I think that this issue may be resolved this way or maybe by using some scaling functions

isobianin, 2020-10-14:
Hi @amchagas !

I have been able to finish the task and I would like you to look at it at my [fork](https://github.com/isobianin/Flypi/tree/master/GUI_2.1). Now there are scrollareas for each tab and they all are nicely incorporated into the current design. 

In the **main.py** I changed one line of code:

if __name__ == "__main__":
... dialog = QtWidgets.QMainWindow() ...

I believe that this is important because this function enables setCentralWidget function which was commented out in flypi_GUI.py

**flypi_GUI.py** was changed substantially but everything remains operational. What I mean is that the graph library is present and corresponding line of code that connects pyqtgraph with the graph is also there. Also, I found it strange that two scrollbars that are in the "camera" tab (horiz and vertical offsets) did not work when clicked on arrows. Now they can be clicked with an increment of 1.

[image]
Fig. 1 - Updated UI

isobianin, 2020-10-16:
Hi @amchagas !

I have tinkered with UI a bit more and now it looks the way it is supposed to look. I have updated backgrounds of scrollareas so they match the previously chosen style (fig. 1)
[image]

isobianin, 2020-10-16:
An additional screenshot of how scrolls work with an example of the "protocol" tab
[image]

amchagas, 2020-10-16:
Hi @isobianin , this looks great,
I am very short on time until Tuesday, but will give it a try then! 
Thanks again for putting time and effort into this!!

amchagas, 2020-12-04:
Sorry, I completely forgot to update this one! The UI looks quite nice and the scrollbars work very well!

amchagas, 2020-12-04:
should probably only close this when we do the PR

isobianin, 2020-12-04:
Hi @amchagas !
Okay, I guess that will do

wexterhack, 2021-01-13:
As I understand this task is closed?

amchagas, 2021-01-18:
Hi @wexterhack yes, this has been dealt with! 
thanks for the interest!

closed 2021-01-18 by amchagas