amchagas/Flypi #72: BOM obsolete parts

neurologic, 2021-04-30:
Hi,
The digikey cart almost fills completely using the current BOM but there are two obsolete parts.
Would you be able to advise on appropriate replacements that fit the specs of the unit and also match the footprint on the gerber?
[image]
https://www.digikey.com/en/products/detail/parallax-inc/452-00007/7791449?s=N4IgTCBcDaICwFYwFoAM7UHZkDkAiIAugL5A
and
https://www.digikey.com/en/products/detail/osram-sylvania-inc/LZ1-10UB00-00U4/5034037

Thank you!

neurologic, 2021-04-30:
I am also having trouble finding a replacement for a backordered part (45 weeks lead time) for component #U4 (IC BRIDGE DRIVER PAR 15MULTIWATT)

amchagas, 2021-05-06:
Hi!
thanks for raising up the obsolete part at DigiKey. I just pushed some changes with a new part that should work. 
For the other parts, dpending on where you are, you could get them  (or very similar ones) from ebay/amazon...
let me know if this solves this issue?

neurologic, 2021-05-07:
Hi Andres. I just saw that push! Thank you!

neurologic, 2021-05-07:
(I am in Connecticut, USA) I used a combination of DigiKey, Mouser, Newark, and Jameco to get all of the parts. A lot was on backorder at each place, but with a combination of all sources  I think I was able to get almost all of the specified parts. I suspect the pandemic is currently a major source effecting supply of component parts? The one item I still was not able to get were the pin housings. But in a pinch, I suppose I am thinking that it looks like I can jump the pins directly to the board from each component?

neurologic, 2021-05-07:
@amchagas, I have another related question. Do you have a recommendation for a "reflective collimator"? I don't see one in the parts list and when I google that term, I can't find something that looks like the silver collar that I see on page 21 of the FlyPi User and Assembly Manual. Should it be something that fits into the "strongLED_holder" printed part? I can't tell from the figure in that assembly pdf, how those pieces fit together and inside of each other. Any help greatly appreciated. Thank you! (If you had any opportunity to have a video chat about the FlyPi set up let me know. I am really excited to set it up for the course I am teaching next semester, but I feel pretty shaky on the details of actually implementing.) I am very very stoked on your design and its applicability.

amchagas, 2021-05-21:
Hi @neurologic ! 
Very sorry about the delay in replying... closing some projects at work plus a small child at home has done me in for the last 2 weeks... 
We can catch up via a video call, sure! Send me an email on a.maia-chagas@sussex.ac.uk (or send me an email address so I can send you a message to arrange things)!

neurologic, 2021-05-26:
Thank you, @amchagas. This sounds great. Sorry for the delay as well. It sounds like we are in some similar situation! I have what still feels like a very new 7month old.
I have all of the parts in now (I think) and am starting to assemble the PCB (rev. 0.6 20171203). And I think I figured my way through the orientation of the 3D parts (and have some notes on edits). I would be happy to try to work on a revised assembly guide while I work through assembling. Meeting to talk about an overview would help a lot since I am pretty novice still at a lot of the electronic circuit principles (but am pretty comfortable with Linux, python, etc so that helps). 
I will email you now.

closed 2021-05-27 by amchagas