amchagas/Flypi #73: peltier module not cooling

rjs360, 2021-05-24:
Hi all,
My build is a v. 2.0, running GUI 2.2. 

Aside from a non-responsive servo module (if anyone has run into this issue please let me know), all seemed to be working fine (all the lighting options are working great) until I tried the peltier module. It heats up fine, e.g. to 35C, but when I try to cool to 15C, it will not cool beyond 25C. I have a heatsink attached to the peltier module, and a fan blowing at it (according to instructions).

Anyone run into this issue? Do I just need more powerful cooling?
Thanks for any insight.
-Ruud

amchagas, 2021-05-27:
Hi @rjs360 !
Cool that you set up a FlyPi, bummer that there are still some glitches. 
Could you share a bit more details on the Peltier, fan and heatsink? 

The reason I ask is that since you have the heatsink and fan, it should work (also considering that most of other things are working as well)...

Another question is how much time does it take to heat from room temp to 35C?

In principle you should not need more powerful cooling...

rjs360, 2021-05-27:
Hi @amchagas, thanks for the reply!  
Yep a couple of bumps in the road but that makes things fun (mostly ;)).

The peltier is a CP85438 (or TEC127050), through DigiKey (102-1682-ND; as listed in the BOM); the fan is a 36x36mm 5V fan, through DigiKey (P19864-ND; as listed in the BOM); the heatsink is the 36x36x12 heatsink (digikey 294-1101-ND) listed in the BOM.  

Heating to 35C is no problem, it takes ~30s to get to 35C from 21C ambient.
I wonder if my thermistor maybe bad?

Additionally, lighting all works fine, servo doesn't, but i worked around that.
-R

rjs360, 2021-05-27:
Make that temp. sensor (AD22100KTZ) instead of thermistor..

amchagas, 2021-05-27:
What PCB board revision do you have?

amchagas, 2021-05-27:
maybe there is another case of board mislabelling?

rjs360, 2021-05-27:
Looks like rev. 0.6
-Ruud

rjs360, 2021-05-28:
Just adding to this: I do get expected values for ambient temperatures using my current wiring. I tried reversing the +/- on the thermistor but that gives me values of about ~150-152 as "ambient" temperatures.  By the way, the latest PCB gerber file (the one that was changed 16 days ago) seems to indicate +S- (from left to right, seen from the top) for the thermistor connector. Did this get changed? My rev. 0.6 board has -S+, but again, reverse wiring does not give expected numbers.  Any thoughts would be welcome, would like to get this sorted out.