amchagas/Flypi #74: PyQt5.sip

neurologic, 2021-05-26:
Hi,
I am working on trying to get the GUI running. I have solved some roadblocks that I have come across so far, but can't get past this one yet after a couple of hours of searching and troubleshooting. So I thought it was time to reach out. Did you do anything special to get PyQt5 installed on your Raspberry Pi with PyQt5.sip as a module? 

I was finally able to get PyQt5 using $ sudo apt-get install python3-pyqt5
But there was not PyQt5.sip module.
I tried making sip from source as PyQt5.sip and that worked... but then there was another PyQt module that was missing and I could not make the dependencies match.

Then I tried installing miniconda thinking that it should have all of these modules. It almost worked. I am still getting an error (now for PyCapsule_GetPointer called with the wrong name). In searching it sounded like downgrading PyQt5 to 5.10.1 would solve that. But I cannot seem to do so. 

Did you have troubles with this? Do you have an environment file that you could share (though I only know how to do that in conda)?

Any insight would be great.

neurologic, 2021-05-26:
Also, I don't know if this is correct or not, but I was going into the GUI_2.2 folder to run the flypi_GUI.py using python3. Given the folders and files available I was surmising that this was correct (as opposed to "sudo python3 ./run.py" instructed in the PDF... it looks like that was for GUI_1.0?)

neurologic, 2021-05-26:
I did try to go into GUI_1.0 and do "python run.py" and a GUI window did generate! Is this what I should still be using? Since there were so many #2 versions and I was working with the newest board version I assumed I should use the 2.x GUIs.

neurologic, 2021-05-27:
I am so happy. I got the GUI_2.2 to run (on my MacOS). I had to change the serial port in main.py. I don't know how to include that as an input variable to the script, but in the near future if I learn how to push some edits to this package I will work on that. 

Now I will try again on my Rpi! Troubleshooting on the Rpi had been frustrating because I have a really really old very slow one and it was hard to learn on. Now that I learned a lot today about the backend I think I might be able to work through some of the issues I was having on the Rpi yesterday. 

[image]

amchagas, 2021-05-27:
Hi, 

Great that you got the GUI to run on another system. And yeah, each operating system treats the serial port in a different way/different nomenclature (in Windows they are COMX - with X being a "random" number for that port, in Linux ttyUSB0 for the Arduino nano and ttyACM0 for the Arduino uno -  no idea for macs.. never used them :P )

About installing things on the Pi, If you have a Raspberry Pi 3 we do provide an image that you can use to create the SD card, which I think is updated and has all things installed, so you should be able to simply run the GUI without having to install anything.. it is here https://www.dropbox.com/s/12o271wj37rqkj6/flypi_image.zip?dl=0

neurologic, 2021-05-27:
Thank you so much, @amchagas . I did see in the documents that there was an image to use. I was avoiding the image in the short term since I had other things running on the RPi and only one SD card currently. 
I don't really know how (but I took some notes)... but I finally got the packages needed and talking nicely with each other. For me it ended up involving downgrading pyqtgraph to 5.11 because PyQt5 install on Pi that I could figure out did not support 5.12 (I think?). And then realizing that I needed to run the code as sudo. 

I was able to hook up and operate the LED ring and the peltier modules! The temperature tracked well and the ring colors all worked. 
There were some edits that seem to be needed that I took note of (for example, some of the +/-/signal notation on the PCB print was mixed up). 

There are some additional edits that I need to make to the code and some questions that I have about getting the optogenetic experiments to work (I am confused which LEDs are expected to be used for this and how). For now it seems I have gotten the hardware and software up and running for the most part. 

I emailed you recently to talk about having a video chat. I have started to try to figure out how to edit the code, but it would be more informative to talk to you about some of these things if you are open to it. We can talk about how I can help revise the assembly PDF, etc as well if you are interested. 

I am also working on a odor delivery system for an electro-antennae-gram experiment in my course. It currently runs off of RaspberryPi through adafruit Crickit. But I would like to translate it to be run by the nano so that I could add a module into your gui for the course (if you are interested in me sharing that with you when I am finished let me know). 

This is a lot of fun. Thank you for putting this all together and sharing!