amchagas/Flypi #75: Arduino Serial Command

neurologic, 2021-05-27:
Hi,
It's me again. I know you are very busy so don't want to create too many 'issues'. I am solving things where I can (and learning a lot). I came across a new issue that I think I know how to get around but am not sure.  At some point, I can learn how to propose/contribute edits via git (I have never done that before and have not been good at following branching protocol, etc in my life; but want to).

Today I am trying to get the Arduino code going so that the Rpi GUI can communicate.  (Yesterday I got GUI_1.0 working on my Rpi but only with the camera module). Today I learned how to communicate with Arduino via Python and serial! (I was learning using pyfirmata). So I think I am understanding the premise behind the FlyPi communication setup between the Rpi and the PCB/Nano. 

So from what I understand currently, you have written arduino code that sets it up to receive commands via python directly (analagous to the pyfirmata 'standard' script that I was loading earlier today onto my nano to control via python). I am working with the most recent script found in the v2 folder called 'serial_test'.

I solved some Library import issues with the adafruit packages by adding via zip downloaded from git directly and found Stephen Cogswell's 'SerialCommand.h' at ```https://github.com/scogswell/ArduinoSerialCommand/blob/master/SerialCommand.h, which I used to 'add library via zip'```.

The #include works for this, but it seems like one of the modules called in your arduino script is not (no longer?) a part of their library. 

ArduinoIDE check fails on:
 ```sCmd.setDefaultHandler(unrecognized);      // Handler for command that isn't matched  (says "What?")```

Looking at Stephen's code, it seems like the command might have changed to 'addDefaultHandler'?
```void addDefaultHandler(void (*function)());    // A handler to call when no valid command received. ```

I edited the Arduino code for addDefaultHandler for now. 
When I did this, the IDE compilation completed.
BUT I do get the following orange text:
```
<my filepath>/Flypi/Arduino/v2.0/serial_test/serial_test.ino: In function 'void setup()':
<my filepath>/Flypi/Arduino/v2.0/serial_test/serial_test.ino:120:38: warning: invalid conversion from 'void (*)(const char*)' to 'void (*)()' [-fpermissive]
   sCmd.addDefaultHandler(unrecognized);      // Handler for command that isn't matched  (says "What?")
                                      ^
In file included from <my filepath>/Flypi/Arduino/v2.0/serial_test/serial_test.ino:31:0:
<my filepath>/Arduino/libraries/ArduinoSerialCommand_master/SerialCommand.h:85:8: note:   initializing argument 1 of 'void SerialCommand::addDefaultHandler(void (*)())'
   void addDefaultHandler(void (*function)());    // A handler to call when no valid command received.
```

Let me know if you think this is an error that I am missing a step somewhere or have the wrong library. 

Also, for the adafruit libraries, my includes ended up looking like this to make it work:

```

#include <Adafruit_NeoPixel.h> // LED Ring

#include <Wire.h> // LED Matrix
//#include <Adafruit_LEDBackpack.h> // LED Matrix
//#include <Adafruit_GFX.h> // LED Matrix

#include <Adafruit_LEDBackpack.h>

#include <Adafruit_GrayOLED.h>
#include <gfxfont.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SPITFT.h>
#include <Adafruit_SPITFT_Macros.h>
```

neurologic, 2021-05-27:
Interesting. For other reasons I just ended up at what is pointed from Stephen's site as 'an alternative':
```https://github.com/kroimon/Arduino-SerialCommand/blob/master/SerialCommand.h```

This one has 'setDefaultHandler'! So this must be the one that you are pointing to/including. 

I just tried that... and it worked without error or orange text! 
So I think I ended up solving this. Hopefully this info is helpful to someone else who might come across the same hiccup.

amchagas, 2021-05-27:
Glad you worked this one out! 
One comment on this one: 

V1.0 of the GUI requires V1.0 of the arduino code, where we are not using the serial library you mention (as far as I remember)...
 
Just in case the communication between the two boards do not work...